oe1wkl/Morserino-32 #3: Additional words

zoglmannk, 2019-04-18:
If there are sufficient resources on the hardware for a larger binary, it would be great to increase the dictionary to include 1,000 words. I am willing to create a pull request that extends the english_words.h file and specifies the top 1,000 common words in the English language.

oe1wkl, 2019-04-20:
I cannot tell right now what the implications would be regarding resources - but I can of course test it. Not sure if 100 might be a bit excessive, not in terms of resources, more along the thinking that this will include some vocabulary that might not be very frequent in common ham radio conversations; but of course this could be edited by hand to have something really useful.

quarksteilchen, 2021-11-22:
these 1000 words wouldn't be mandatory as well. a little option to "Enable extended word list" would be a solution for more words :-)

oe1wkl, 2021-12-02:
Kurt, I am not sure if the memory would be sufficient, but its probably worth a try. Currently there are 370 words used  (I think). Where are you taking the words from? Ideally we would get those from real ragchew QSOs, not sure if this is doable…

Willi

closed 2026-09-01 by oe1wkl