oe1wkl/Morserino-32 #8: WPM calculation incorrect?

kf9up, 2019-09-20:
I noticed the effective Wpm was always 1 less than the Wpm  (when using the default of 3 for inter character spacing and 7 for interword spacing).  I don't believe this is correct.

The documentation says the "interword space will always be at least 4 dit lengths longer that the character spacing", which makes sense.  However, the code in updateTimings() adds 6 to the character spacing:

    interWordSpace = _max(p_interWordSpace * ditLength, (p_interCharSpace+6)*ditLength);

This makes the calculation of "effWpm" be 1 less than what I think it should be.  I think it should read "+4" not "+6".  You can also remove 1 multiplication with:

    interWordSpace = _max(p_interWordSpace, p_interCharSpace+4) * ditLength;

Hope that all makes sense.

oe1wkl, 2019-09-20:
> I noticed the effective Wpm was always 1 less than the Wpm (when using the default of 3 for inter character spacing and 7 for interword spacing). I don't believe this is correct.
>
> The documentation [...]
You're absolutely correct, that is a mistake. Will be corrected in the next release (due in about a week to 10 days….)

Willi

aaaaalbert, 2021-07-13:
This seems to be fixed since around 3f9b437304f3c2863269236d8e7834886bd2f931. I guess this issue can be closed :-)

oe1wkl, 2021-07-14:
True, has been fixed a while ago...

closed 2021-07-14 by oe1wkl