oe1wkl/Morserino-32 #11: Any need of increased support for straight keys?

Karethoth, 2019-10-26:
I've been fooling around the code toda, and so far I've gotten the echo trainer to support straight key without too many apparent problems. I've pushed the changes to my fork. If there is any interest in them, anyone is free to use the changes as they see fit.

The major changes I've made so far:
- Added "Straight" keyer mode, works wherever echo trainer is used (koch trainer, echo trainer)
- "<err>"s and whatever came before them in the echo trainer response get ignored (I noticed this made me less frightful of making mistakes (wrong amount of dits) when using paddles)

What doesn't work:
- LoRa TX when in straight key mode
- Really short dits don't always get registered
- May have missed other stuff since there are so many features to test

If there is interest for increased straight key support, I'll gladly iron out the little problems that the fork currently has. Just curious to know if there is need for this.

NickClark, 2019-12-02:
Just learning CW here, so excuse my naivety, but I'm planing on trying a bug with this... does straight key functionality affect that? Does it already support it? Because if this is related to bug use, then *yes*, I'm interested :D

oe1wkl, 2019-12-03:
Nicholas,

You can use a bug in the same way as a straight key right now - the only supported functionality is the decoder. So you can use the Morserino-32 to decode what you are keying with your bug, and show this on its display. So that you can check if your keying is „decodable“.

Be aware that with a bug, depending on its setting, the relationship between dots and dashes can be pretty far from the norm (1:3) - this will have an effect on the ability to decode, i.e. if the relationship is too far off, the Morserino will have difficulties to decode (but human operates will eventually run into the same problem…)

Other Morserino features (like the echo trainer) do currently not support a straight key (or a bug), but only paddles.

73

Willi

asdil12, 2020-05-08:
I guess this got greatly inproved with firmware v3.0.

closed 2021-07-14 by oe1wkl