oe1wkl/Morserino-32 #14: Sound output disabled after altering settings

Koppo82, 2020-02-26:
Version 2.3

1. Go to:
2. CW Generator
3. Settings: change a setting (e.g. quickstart ON)
4. Return to CW generator

Result:
No sound is heard anymore. Device needs to be restarted. 

Expected results:
Settings are saved without the device needed to be restarted.

oe1wkl, 2020-02-26:
I cannot verify this behavior. I followed your steps, and I still get sound output.
I looked at 2 different circumstances:
1. stop CD Generator with a paddle press, and then change the setting; restart with a paddle press
2. Do not stop, go to settings immediately while CW Generator is running; I have to restart with paddle press.

In both cases I have sound output as expected.

Is there anything you did differently?

Willi

Koppo82, 2020-03-07:
Hi Willi, 

sry for late reply.
I can reproduce it like this:
Open CW Generator and select cw abbr. During generation open settings and change "repeat each word" to 2x. Go back. Sound dissappears. Switching to another mode reinitiates the sound. Switching back to cw generator resolves the issue completely.

oe1wkl, 2020-03-20:
Hi!

I did a lot of tests, was never able to replicate this behavior. This is very strange. Could it be a timing issue? At slow speed I even tried to interrupt CW generation a) while the sound was playing, b) during a pause; the effect never manifested itself.

I have no idea what could cause this in your case. Also did not hear about such behavior from other users.

73
Willi