oe1wkl/Morserino-32 #16: Echo Trainer File Upload Stops when Echo Repeats are not Unlimited

jsurls, 2020-04-04:
While practicing with the echo trainer with an uploaded practice file, if echo repeats are not "unlimited" the echo trainer comes to a stop after the failed Nth repeat and does not continue unless you exit and start again. This does not occur in the other built in modes.

Setting to reproduce:
```
Max # of Words = Unlimited
Randomize File = Off
Each Word 2x = Off
Echo Repeats = 0 (or any number other than Unlimited)

```

To recreate set Echo Repeats to 1 and upload a file with just the letter "a" on one line. Let it fail twice (you have 1 repeat) and it will stop supplying words.

A funny side effect of this is partly what I wanted as a feature: 
Track number of failures (or max them out) forcing the user to attempt to try and have a perfect run (or run with no more than N failures).

oe1wkl, 2020-04-04:
Joey,

That's weird. Could it be that this happens only with the very specific one word / one line file? 
A first quick check with a longer file that's currently residing on my Morserino did not show this behaviour. Will check tomorrow with the one word file....

Willi

jsurls, 2020-04-04:
My original file was the following:

```
B
Z
V
Q
H
G
2
/
,
J
```

I was trying to practice certain letters and had it set to 0 incorrect. It mostly worked and eventually stopped. Not sure why. I then updated to the latests 2.4 and ran into the same issue even after uploading a new file. I then tried to narrow to the "simplest" version of the problem that I could consistently reproduce.

I looked through the code and nothing immediately stood out. I even considered maybe the file didn't have a newline at the end and perhaps the file player is falling out of bounds by chance.

To make sure I'm resetting state (at least I think I am..) I am turning the switch to off then on again rather than sleep. Let me know if there's another way you'd like me to try it. I can also upload a video if necessary.

oe1wkl, 2020-04-05:
It seems that it always stops at the end of the file. Not sure yet if there are other parameters that have an influence, need to check that. Will fix that hopefully in the next release…

73
Willi

oe1wkl, 2020-04-06:
Joey,
I was able to find out what went wrong.. It always happened at the „end of file“ condition, in echo trainer mode only.
Will be fixed at the next release.

Willi