oe1wkl/Morserino-32 #17: WiFi TRX mode

asdil12, 2020-04-26:
Especially due to the current circumstances it would be nice to have a WiFI TRX mode that works analog to the LoRa TRX.
This would allow distant users to connect their Morserinos directly via WiFi (using a VPN or Router port redirections).

The Morserino could send simple UDP packages using the same package header/format as it currently does for LoRa.
The existing WifFi settings could be used, but extended by a Peer entry for the IP address of the Peer Morserino.

This are my ideas how to begin:
```patch
--- ./morse_3_v2.4.ino	2020-04-26 20:44:04.662410857 +0200
+++ ./morse_3_v2.4.ino	2020-04-26 21:10:35.903208571 +0200
@@ -57,6 +57,7 @@
 #include <SPI.h>           // library for SPI interface
 #include <LoRa.h>          // library for LoRa transceiver
 #include <WiFi.h>          // basic WiFi functionality
+#include <WiFiUdp.h>       // UDP lib for WiFi TRX
 #include <WebServer.h>     // simple web sever
 #include <ESPmDNS.h>       // DNS functionality
 #include <WiFiClient.h>    //WiFi clinet library
@@ -533,6 +534,7 @@
   uint8_t p_menuPtr = 1;                      // current position of menu
   String  p_wlanSSID = "";                    // SSID for connecting to the Internet
   String  p_wlanPassword = "";                // password for connecting to WiFi router
+  String  p_wlanTRXPeer = "";                 // peer Morserino for WiFI TRX
   uint32_t p_fileWordPointer = 0;             // remember how far we have read the file in player mode / reset when loading new file         
   uint8_t p_promptPause = 2;                  // in echoTrainer mode, length of pause before we send next word; multiplied by interWordSpace
   uint8_t p_tLeft = 20;                       // threshold for left paddle
@@ -831,6 +833,8 @@
 
 WebServer server(80);    // Create a webserver object that listens for HTTP request on port 80
 
+WifiUDP udp;             // Create a udp socket for wifi trx
+
 File fsUploadFile;              // a File object to temporarily store the received file
 
 String getContentType(String filename); // convert the file extension to the MIME type
@@ -852,6 +856,7 @@
                 "<label for='ssid'>SSID of WiFi network?</label>"
                 "<input name='ssid' id='ssid' ></div> <div>"
                 "<label for='pw'>WiFi Password?</label> <input name='pw' id='pw'>"
+                "<label for='pw'>WiFi TRX Peer?</label> <input name='trxpeer' id='trxpw'>"
                 "</div><div><button>Submit</button></div></form></body></html>";
 
 
@@ -5077,10 +5082,12 @@
     server.send(200, "text/html", "Wifi Info updated - now restarting Morserino-32...");
     p_wlanSSID = server.arg("ssid");
     p_wlanPassword = server.arg("pw");
+    p_wlanTRXPeer = server.arg("trxpeer");
     //DEBUG("SSID: " + String(p_wlanSSID) + " Password: " + String(p_wlanPassword));
     pref.begin("morserino", false);             // open the namespace as read/write
     pref.putString("wlanSSID", p_wlanSSID);
     pref.putString("wlanPassword", p_wlanPassword);
+    pref.putString("wlanTRXPeer", p_wlanTRXPeer);
     pref.end();
     
     ESP.restart();
@@ -5539,6 +5546,7 @@
 
     p_wlanSSID = pref.getString("wlanSSID");
     p_wlanPassword = pref.getString("wlanPassword");
+    p_wlanTRXPeer = pref.getString("wlanTRXPeer");
     p_lcwoKochSeq = pref.getBool("lcwoKochSeq");
     p_quickStart = pref.getBool("quickStart");
```

oe1wkl, 2020-04-27:
That’s an interesting concept. As downsides I see:
- you have to enter the IP address of your QSO partner manually
- it will be difficult to cross firewalls / IP masquerading; people would have to define ports in there routers for that.

That’s why all chat protocols use some mediation service on the Internet, in order to address these two points (and probably a few more). All in all that would create a formidable project, I guess.

Easier is to use iCW (CW over Internet), which uses an established service (Mumble). Downside is the Morserino is not enough, you need a computer as an interface…

Willi

asdil12, 2020-04-28:
See Pull Request #18

PR #18 (2020-05-01): Add Wifi Trx mode
Fixes #17

Todo:
- [x] Rename functions to more general names
- [x] Shutdown wifi after usage
- [x] Implement receiving
- [x] Work on FIXMEs
commit 13990b37 (2020-05-01): Add Wifi Trx mode

Fixes #17
closed 2020-05-01 by oe1wkl