oe1wkl/Morserino-32 #22: IP 0.0.0.0 can be shown on firmware update screen

asdil12, 2020-05-08:
When opening "check wifi", leaving it via red button and then opening the firmware update, only "IP: 0.0.0.0" is shown.

ttomasini, 2020-05-08:
Found the issue: We need to disconnect every time we exit a wifi related function. Otherwise we are calling `WiFi.begin(...)` two times consecutively and that causes the issue. The Arduino WiFi lib does not seem to intercept this edge case.

I'll try to send a patch tomorrow.

oe1wkl, 2020-05-10:
I fixed the issue and will upload a patched version tomorrow.

Willi

oe1wkl, 2020-05-11:
Has been resolved.

closed 2020-05-11 by oe1wkl

ttomasini, 2020-05-11:
This fixes the issue with the IP addresses but I'm not entirely sure if that really shuts of the WiFi itself. I would rather replace `WiFi.disconnect(...)` with `WiFi.mode(WIFI_MODE_NULL);` – just to make sure.

Anyway, thanks for the quick fix!

oe1wkl, 2020-05-11:
The ’true’ parameter should do exactly that ( shutting down WiFi), according to teh documentation ;-) no idea if it really does it.

Willi

ttomasini, 2020-05-11:
Ah, I see. But then we should replace all of the other `WiFi.mode(WIFI_MODE_NULL);` with `WiFi.disconnect(true, false)` to improve consistency. :)