oe1wkl/Morserino-32 #28: RFE: protected way for WiFi password set-up

yarda, 2020-05-23:
Today, it's really easy and cheap to setup RF/WiFi sniffer, so it pose real security risk to send the WiFi password over the air in the plaintext. Even it's small window during which the password is sent it's still possible to catch it and compromise the host network. It's not much problem in rural areas, but in the city, especially in the height spot/building there can thousands of people in the radio range.

Could Morserino setup the WPA2 protected AP for the initial WiFi setup? The low security approach is to use hardcoded password like "morserino", high security approach is to generate random password and show it on the display. Or could the password be send over HTTPS? It would lower the risk of automated full-time 24/365 sniffers and possible offline analysis of the captured data - realtime man-in-the-middle attack is unlikely to happen in the small time window. Or could be there an option to set the password through the USB? For security concerned people like me? :)

kitesurfer1404, 2021-06-06:
I ordered my morserino some days ago and thought the same when watching the videos. In the past I used the ESP8266 a bit and I'm some kind of familiar with this topic.
As far I can tell, a password protected AP mode is already prepared in the code:
There is a place for a hardcoded password:
https://github.com/oe1wkl/Morserino-32/blob/273c0642df671fc362f9d16e23e84f3f0afb5a39/Software/src/Version%204/MorseWiFi.cpp#L35

I think using the password here: 
https://github.com/oe1wkl/Morserino-32/blob/273c0642df671fc362f9d16e23e84f3f0afb5a39/Software/src/Version%204/MorseWiFi.cpp#L361

WiFi.softAP(ssid, password);

would fix this.

I'm interested in this as well. So I try to give it a look as soon as my hardware arrives. 
I find the suggested solution with the random password shown on the display interesting. Only change I would implement is "pseudo random", so the password for each morserino is different but always the same for your own device. So you don't have to change the password each time you want to connect.

bensteinberg, 2022-05-07:
Another approach to this would be to allow SSID and password entry using the paddles. This might be a challenge for beginners, of course. I think you'd need to simulate shift, backspace, space, and tab keys using the encoder and red button, and disregard any space between characters entered with the paddle.

bensteinberg, 2022-05-07:
(It would probably be simpler to implement (and use) a text entry method using the encoder to select and set characters.)