oe1wkl/Morserino-32 #34: Koch sequence not memorized !

philgerm, 2020-10-07:
On version 3.02  the Koch sequence LCWO can't memorized on snapshoot,  after recall a snapshot the Koch sequence go to M32/ JLMC. the code looks correct in Write and readPreferences !

rainerrudigier, 2020-12-11:
looks like that in the writePreferences function something got mixed up .. 

the lcwoKochSeq is only saved to the morserino savestate - but never to a snapshot

    if (morserino) {
        if (MorsePreferences::kochFilter != pref.getUChar("kochFilter")) {
            pref.putUChar("kochFilter", MorsePreferences::kochFilter);
            if (!MorsePreferences::useCustomCharSet)                          // we update these only if we do not use a custom character set!
              koch.setup();
        }
    
        if (MorsePreferences::lcwoKochSeq != pref.getBool("lcwoKochSeq")) {
            pref.putBool("lcwoKochSeq", MorsePreferences::lcwoKochSeq);
            koch.setKochChars(MorsePreferences::lcwoKochSeq);
            if (!MorsePreferences::useCustomCharSet)
                koch.setup();
        }
    }

it should be changed to this (like in v2.4) i think:

    if (morserino) {
        if (MorsePreferences::kochFilter != pref.getUChar("kochFilter")) {
            pref.putUChar("kochFilter", MorsePreferences::kochFilter);
            if (!MorsePreferences::useCustomCharSet)                          // we update these only if we do not use a custom character set!
              koch.setup();
        }
    }
    if (MorsePreferences::lcwoKochSeq != pref.getBool("lcwoKochSeq")) {
        pref.putBool("lcwoKochSeq", MorsePreferences::lcwoKochSeq);
        **if (morserino) {
            koch.setKochChars(MorsePreferences::lcwoKochSeq);
            if (!MorsePreferences::useCustomCharSet)
                koch.setup();
        }**
    }

tried it out yesterday evening and it seems to work ..

oe1wkl, 2020-12-11:
Thanks for pointing this out.
I am, currently preparing v. 4.0 (which will also support the new edition of M32 with the newer Heltec Module version 2.1), and I will incorporate that bug fix into that version.

Willi

rainerrudigier, 2020-12-11:
thanks

73 de OE9RIR

closed 2026-08-31 by oe1wkl