oe1wkl/Morserino-32 #37: Please port to the Raspberry Pi platform.

Supermagnum, 2021-01-04:
I think that a shield that plugs into the Raspberry Pi family should be developed.

oe1wkl, 2021-01-04:
I do not have the skills or resources for that. But this is open source software, so everybody is invited to do so, if tehy feel there is a benefit in it.

Supermagnum, 2021-01-06:
> I do not have the skills or resources for that. But this is open source software, so everybody is invited to do so, if tehy feel there is a benefit in it.

Maybe this is a viable option:
https://github.com/me-no-dev/RasPiArduino

chrisdebian, 2025-07-07:
I don't know any of the details of the upcoming 3rd Generation device, but given the reply above, can we conclude that unless the software is forked, this won't be on the future roadmap for this project? 

Chris

oe1wkl, 2025-07-09:
Yes, Morserino will stay on the ESP32 platform.

closed 2025-07-09 by oe1wkl