oe1wkl/Morserino-32 #43: Git Clone Fails on Windows: Invalid path 'Software/UDP Chat Server by SP9WPN /README.MD'

K6TJO, 2021-08-22:
Cloning into 'Morserino-32'...
remote: Enumerating objects: 1181, done.
remote: Counting objects: 100% (63/63), done.
remote: Compressing objects: 100% (46/46), done.
remote: Total 1181 (delta 31), reused 35 (delta 17), pack-reused 1118 eceiving objects: 100% (1181/1181), 112.65 MiB | 22.21 MiB/s
Receiving objects: 100% (1181/1181), 119.31 MiB | 21.79 MiB/s, done.
Resolving deltas: 100% (710/710), done.
error: invalid path 'Software/UDP Chat Server by SP9WPN /README.MD'
fatal: unable to checkout working tree
warning: Clone succeeded, but checkout failed.
You can inspect what was checked out with 'git status'
and retry with 'git restore --source=HEAD :/'

oe1wkl, 2021-08-29:
I have no idea what this means or what the reason is for this error message… :-(

Willi

k6run, 2021-08-29:
Your OS doesn't like the filename. You must be on something other than
Linux.  You should see all files except that one README.

oe1wkl, 2021-08-29:
I'm on OSX and I see all file names and files… 
This how it looks in my file system:

And this how it looks on GitHub:

Cannot see anything suspicious...

Willi

k6run, 2021-08-29:
I would guess it's the space at the end of the directory name that it
doesn't like

SP9WPN /

I'll have to give it a try on my mac to verify.

K6TJO, 2021-08-31:
I am on Windows.

backbaseLT, 2021-08-31:
Remove the space and those who use Windows won't get this error

K6TJO, 2021-09-01:
Works now!  Thanks

closed 2025-07-09 by oe1wkl