oe1wkl/Morserino-32 #48: Start Echo Trainer with Straight Key

garrettben, 2021-12-08:
Currently the echo trainer will only start with wither an iambic key or one of the PCB paddles, and not with the straight key alone.  Once I have started either the koch or standard echo trainer with a paddle, it works fine with a straight key.  This is with the current firmware m32_v4.1.ino.wifi_lora_32_V2.bin

oe1wkl, 2021-12-08:
You can also start it by pushing  the black encoder knob…

Willi

jsurls, 2022-03-14:
I just downloaded the update and also anticipated my bug being able to start the echo trainer in the same way my paddle had done in the past. Quickly figured out the black encoder knob is a workaround, however it's a bit inconvenient.

+1 on seeing this improvement in future revisions

closed 2026-08-31 by oe1wkl