oe1wkl/Morserino-32 #54: Unpexected restarts in 'Adaptive Random' mode

DO1GDO, 2022-03-08:
**Problem:**

Morserino is restarting unexpectedly and randomly after some characters have been keyed in 'Adaptive Random' mode.

**Reproduce:**

Select the last koch lesson ":" and start adaptive random mode. Restart should happen within 100 keyed characters.
Please confirm.

**Cause:**

Unclear. Given that it only happened on my devices with the last koch lesson, i would suspect a out-of-bounds array access. But i did not find it until now.

One time, i received the following on USB-serial-output:
```
ia3 >ia3 OK
Guru Meditation Error: Core  1 panic'ed (Unhandled debug exception)
Debug exception reason: Stack canary watchpoint triggered (loopTask) 
Core 1 register dump:
PC      : 0x4010b36d  PS      : 0x00060336  A0      : 0x800eac01  A1      : 0x3ffb1e20  
A2      : 0x3ffb1e50  A3      : 0x3ffd21e0  A4      : 0x00000003  A5      : 0x0000002f  
A6      : 0x000000a5  A7      : 0x3ffb1fb0  A8      : 0x00000000  A9      : 0x3ffb1e00  
A10     : 0x3ffb0061  A11     : 0x3f402d79  A12     : 0x00000000  A13     : 0x00000003  
A14     : 0x3ffb1f0c  A15     : 0x0000000a  SAR     : 0x0000000a  EXCCAUSE: 0x00000001  
EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xffffffff  

Backtrace: 0x4010b36d:0x3ffb1e20 0x400eabfe:0x3ffb1e40 0x400eacb2:0x3ffb1e80 0x400f8a8f:0x3ffb1ec0
0x400fbfb6:0x3ffb1ef0 0x400fc1fb:0x3ffb1f40 0x400fc76f:0x3ffb1f80 0x4010b729:0x3ffb1fb0 0x40089ccd:0x3ffb1fd0
```
**Possible Solution:**

I did not found a out-out-bounds-array-access until now. However, reducing heap memory allocations and avoiding string reallocations, especially on the stack, seems to solve the problem.

I created a branch here and committed some enhancements:
https://github.com/DO1GDO/Morserino-32/tree/fix/adapt-rand-memory

@oe1wkl: 
Still working on it to improve the behaviour. However, some refactoring is still to be done. Problem does not happen any more with the current commits in the fix-branch, so we could review and test those commits for a merge ... or i could try to do some further refactoring before merging ... let me know your preference please.

I am a bit confused about custom-chars handling ... is it a still active feature or was it removed at some point in the past? (sorry, didn't try all morserino options and features until now ...

oe1wkl, 2022-03-08:
Custom characters is still supported. It essentially allows the user to upload a file containing the characters they want to use, and then this set of characters is being used instead of the Koch chars up to a specific lesson.
As the restart seems to occur only at the last character. the number of users affected by it is probably small, so we can take a few days to explore this in some more depth...

DO1GDO, 2022-03-09:
Reproduce:

1. Select the last koch lesson 50 ":"
2. Start adaptive mode with 3 chars
3. Enter a wrong char as last char e.g. "9qo" is asked, you enter "9qm"
4. Enter the correct chars with the second attempt (in this example "9qo")

=> Morserion will restart

Method echoTrainerEval() will be completed successfully before the restart happens (checked with additional printOnDisplay() at the end of the method).

Still no luck with any obvious array-out-of-bounds-access.

DO1GDO, 2022-03-12:
Found my mistake and fixed in PR #55 
Thanks for you patience :)

closed 2022-03-20 by DO1GDO