oe1wkl/Morserino-32 #57: File Player plays some strange html-headers ...

slashTPA, 2022-05-07:
Hi,

whatever TXT file I upload (also the given example files from github) - once I start the File Player it always plays the same: Meta data of a html file. How can I get rid of that effect? I'm on latest SW version and did try out reset already as well.

Thx.

oe1wkl, 2022-05-07:
Did you by chance ever try to upload a file larger than 1 MB? In this case the file system might have been corrupted... 
You should never upload html files, as the file player can only play ASCII text files. 
If the file system is corrupt, you need to reflash the firmware via USB with the erase option. 

Willi

slashTPA, 2022-05-08:
I did'nt upload html-files for sure. regarding the filesize - I uploaded the "https://github.com/oe1wkl/Morserino-32/blob/master/Practice%20Files/checkpartial_contest_calls_random.txt" as my largest file ever... 

Please share **HOW** will I do the "reflash the firmware via USB with the erase option." - what tools/program do I need for it? 
Without the File Player option, the morserino is useless for me now...and I would need to fix it urgently

Thx, 
Tom

oe1wkl, 2022-10-11:
Re. Updating the firmware, look at https://github.com/joewittmer/Morserino-32-Firmware-Updater <https://github.com/joewittmer/Morserino-32-Firmware-Updater> and Appendix 4 of the Morserino user manual.

If your Morserino plays HTML code, some HTML code apparently has been uploaded to it. Check the txt file you uploaded with a text editor, maybe the download from GitHub was not done correctly and some html code has crept in during that step. Could also mean the file became a lot longer, maybe over 1000 kB, in that process (you can also check the file size of the file you uploaded).

73

Willi

closed 2024-10-27 by oe1wkl