oe1wkl/Morserino-32 #62: Morserino 32 with heltec V2 board no display after updating to 4.4

JohnCanty, 2022-10-31:
Updated via the web interface, V4.4 successfully installed as guessed by seeing the quick flash of Lora init successful. I then attempted to update via the command line update utility and I get the same problem. Success reported by the utility, flash of the lora init, but no display. When I rotate the knob it makes noise, I can even start one of the trainers by selecting whatever is presented. Any troubleshooting tips, or is this a software bug?

JohnCanty, 2022-10-31:
Additional bits of information. V4.3.1 works. Maybe this helps.

oe1wkl, 2022-10-31:
Seems to be a hardware problem, at least nothing like that has been reported so far. Try to install an earlier version using USB (you can downgrade), and make sure the binary file is not corrupt (verify the MD5 checksum). WilliVon meinem Mobiltelefon gesendet

oe1wkl, 2022-10-31:
Did you verify the MD5 checksum? WilliVon meinem Mobiltelefon gesendet

JohnCanty, 2022-10-31:
My MD5 : 56411610aea2075dedd7ed0ad7c07745
Github MD5: 56411610aea2075dedd7ed0ad7c07745

At first glance these look the same. Didn't super inspect it until now. But yeah the same.

JohnCanty, 2023-04-03:
Just did an update to the 4.5.1 that was released 3 weeks ago. Still the same result. What changed between 4.3.1 and 4.4?

JohnCanty, 2023-04-03:
Found this in the release notes. "You can set the brightness of the display, by double-clicking the RED button. There are 5 levels of brightness. Each double-click reduces the brightness a bit, when the lowest level has been reached a double click resets the display to full brightness again."

JohnCanty, 2023-04-03:
This was added, along with a morsemenu line: Switch case statement with 2 states -1, and 2

void MorseOutput::decreaseBrightness() {
    switch (MorsePreferences::oledBrightness) {
      case 255: 
                MorsePreferences::oledBrightness = 127;
                break;
      case 127:
                MorsePreferences::oledBrightness = 63;
                break;
      case 63:
                MorsePreferences::oledBrightness = 28;
                break;
      case 28:
                MorsePreferences::oledBrightness = 9;
                break;
      default:
                MorsePreferences::oledBrightness = 255;
                break;
    }
    Heltec.display -> setBrightness(MorsePreferences::oledBrightness);
}

oe1wkl, 2023-04-03:
John,

As so many units are out without this problem (both delivered with newer versions, or upgraded), I have to assume it is basically some hardware fault. I am going to ship you a new Heltec free of charge, hoping that this will get rid of the problem.
Once you have received the new module, and verified everything is OK, I would be grateful (if it is not too costly) if you could send me back the faulty Heltec module.

73

Willi

JohnCanty, 2023-04-03:
Yeah I can certainly send you back my Heltec module. I also want to find out what exactly is happening. because during the startup I am seeing a message flash across the screen quickly "LoRa initialized" I think is what it says. So I know the screen is working, and even that this variant of firmware can write to it. So, my plan was to dig through the firmware between 4.3 and 4.4 and sort out the offending line or lines of code. :)

oe1wkl, 2023-04-04:
New Heltec module has been mailed today.

73

Willi

JohnCanty, 2023-04-05:
Awesome, thank you. After looking through the code, a lot changed between 4.3.1 and 4.4. I do have a Heltec V3 here, I tried to compile the .ino for that and the way the libraries are called have changed a bit. Is there a plan for upgrading to the Heltec V3? If so I can help beta test.

oe1wkl, 2023-04-05:
Eventually a change to V.3 will be necessary; I am a bit reluctant right now, as the binaries will not be compatible, so two versions need to be maintained in parallel. Not sure yet if hardware changes will also be necessary (or desirable for other reasons).

Willi

JohnCanty, 2023-06-01:
Hey Will,

I sent the old helltec board back to you via slow and inexpensive mail. I don't know if you have got it yet. On the board I had the previous v4.3 on it that worked. You should find updating it to 4.4 causes the display to not show anything. Also, I Installed the new board and things are working well. Thank You!!

oe1wkl, 2023-06-02:
John,

I got the module last week. So far I had only the opportunity for a quick check, and it confirms your findings.
I noticed that the display is rather dim anyway… my guess right now is that this is a faulty display. Will have to investigate further, how this became manifest with the version that introduced changing the brightness of the display.

Willi

closed 2026-08-31 by oe1wkl