oe1wkl/Morserino-32 #67: Make Sound from Speaker less harsh

GavinAndrews, 2023-01-04:
Creating an issue for discussion...

The speaker sound to me is very harsh.  I assume this is because it is fed from a PWM square wave and has abrupt starts and stops causing harmonics at the end.

Could this instead be fed from the ESP32's cosine wave generator and use the attenuation options 1/2, 1/4, 1/8 at the ends for optional decay at the start and ends?  This would be 8-bit DDS might be enough?

Has this been tried?  I'm tempted to give it a go.

oe1wkl, 2023-01-04:
While this would be a nice undertaking, I think it is not possible with current M32 hardware (maybe an idea for a next generation Morserino?): You need one of the two DACs of the ESP32, which are on pins 25 and 26. Pin 25 is currently used as an output for keying an external transmitter, and Pin 26 is used internally within the Heltec module to control the LoRa module.

Willi

GavinAndrews, 2023-01-04:
WiFi_Kit_series-master\esp32\libraries\LoraWan102\src\driver\board-config.h

#define RADIO_DIO_0                                26

Ah.  That's a problem.

The only easy bodge would seem to be to swap GPIO23 and GPIO25 and perhaps remove R14 on the Heltec to avoid any interference from the LED.   Nasty.

[image]

GavinAndrews, 2023-01-05:
OK... so here's my plan... not for the faint hearted...

I intend to lift R8 and R10, put the resistors vertical on one pad and then wire patch wires to the other pins; swapping Pin23 and Pin25...  I cannot do that currently since obviously the sound is still currently being generated in Pin23.

In the interim I have moved the keyerPin to pin 13; freeing up pin 25.

I wrote a small snippet of code to drive the now free pin 25...

[image]

Looks about right.

Now my analogue electronics is a bit weak... this signal obviously has a bias to it... What's the best way to connect to T2?

GavinAndrews, 2023-01-05:
If I ignore T1... probably short it...

[image]

GavinAndrews, 2023-01-06:
Incidentally the BC817 which I see in the BOM looks like a strange decision for an AND gate.  A 2N7002 might be a better choice?

oe1wkl, 2023-01-06:
Why? Von meinem Mobiltelefon gesendet

GavinAndrews, 2023-01-08:
I haven't given up but I can see I did make an error with respect to remapping the keyerPin to 13... this is already used by the battery voltage sense.

I'm now thinking of using I2S but would need 3 pins... 22 & 23... but still need one more.

closed 2026-08-31 by oe1wkl