oe1wkl/Morserino-32 #68: Wifi configuration is reset to very old config every week

cdaller, 2023-02-06:
Steps 
* I setup 2 different wifis using two different SSIDs and using cq.morserino.info for both
* everything works, with both wifis
* a week later, the wifi configuration is reset to the first wifi configuration I ever did (weeks ago): only one wifi is configured and no cq.morserino.info 

Strange thing is, that when I configure
SSID mobile as first wifi and SSID home as second wifi
after the wifi configuration resets itself, there is only wifi home left (which was the first wifi I ever configured weeks ago).

This behaviour was observed now 3 times in a row. Every week on my cw school day, the configuration is reset.

I do change the snapshots in between and use the serial communication - but I cannot say if this is the culprit!

oe1wkl, 2023-02-07:
Looks like a bug (somewhere where the parameters are being stored…) I will go and hunt for it….

Willi

oe1wkl, 2023-02-17:
I tried to find out more about that issue, checked the code etc.
But I cannot find any reason that could be made responsible for this behavior…. No idea what is going on here.
Serial communication is unlikely, as that does not write to the flash storage. I also cannot believe that restoring snapshots could delete the WiFi configs.

Maybe we need closer observation (e.g.: whenever you go to a different snapshot, can you check WiFi Select if anything has changed there?)

Willi

cdaller, 2023-02-17:
Hello Willi,

I can reproduce the behaviour:
* Wifi Config: set 1:Wifi A and 2:Wifi B
* morserino restarts
* wifi select shows 1:Wifi A and 2:Wifi B
* I select Wifi B
* Double Click on black button, load a snaphot (even the same that was loaded before)
* Wifi Select shows only 1:Wifi B (as first Wifi), no other Wifis are shown anymore

When I call Wifi Configure then: only 1:Wifi B is shown

Wifi B was the first Wifi I ever configured on slot 1. This seems to come back whenever I change snapshots.

This happens always, independently if I switch to snapshot 1 or 5 - no difference

I could help to debug if you want to add some serial output or something like this.

cdaller, 2023-02-17:
I just updated from verson 4.3.1 to 4.5 but the problem is still existing

oe1wkl, 2023-02-17:
Now I think I know what happens:

- snapshots store almost all settings, so they also store the WiFi Config that was configured at the time
- when you restore a snapshot you are restoring the WiFi config that was active back then

Possible solutions:
- you make sure you have all WiFi configs when you store a snapshot
Or, if this is counterintuitive,
- we could change the code that snapshots do not store WiFi configs (and so nothing is changed when you recall a snapshot).

The thinking back then was to use snapshots for different Wifi environments (snapshots were introduced before there were three different WiFi configs).

What does make more sense? Do we really need snapshots for memorizing different WiFi configs, now that we can have three configs anyway? Are three settings enough (I would think so for most people: home, workplace, mobile WiFi). Of course we have also the option to create more than three configs …

Willi

cdaller, 2023-02-17:
Yes, I thought it is something like this. I think, you should not store the wifi configs with the snapshots but separately from them. Otherwise I need to configure my wifis for all 6 snapshots. As you can save multiple wifis, there is no need to store them in the snapshots. And I think 3 of them are enough. But if it easy to extend, provide 5 wifis to save - this should be enough for all of us I would think.

Thanks a lot for your time to find/fix this problem of mine (and probably of others as well).

73 de OE6CHD Christof

closed 2026-08-31 by oe1wkl