oe1wkl/Morserino-32 #73: Will not connect to wifi

VE3JXX, 2023-04-30:
Built a morserino, works great except can't connect to wifi, was told to upgrade to latest software V4.5.2 this was done. config M2 then connect to m2 on computer, enter wifi name and password...M2 resets then I check wifi and evry time it says "No Wifi"
My computer sees "Morserino" and M2 sees my router name, they just won't connect...any suggestions??
Thanks in advance DE John VE3JXX

oe1wkl, 2023-04-30:
Does you router support 2.4 GHz Wifi? Heltec modules cannot do 5 GHz WiFi…

Sometimes it also helps to try „Check Wifi“ 2 or 3 times…

W

VE3JXX, 2023-04-30:
Yes forgot to say my wifi is 2.4 GHz have tried connecting 10-15 times, even had the LICW group assisting, but no luck.
John

oe1wkl, 2023-04-30:
Strange and unusual.
Is there a setting in your router that might prevent connection, like a setting that allows only certain MAC addresses?
What make is your WiFi router? Could you possibly try to connect to another WiFi router (eg. Within your family or at a friend's place)?

Willi

VE3JXX, 2023-04-30:
Router is a "Bell" (Bell Canada) network, I'm planning on going to a ham buddy's place this week to retry on his network....

closed 2026-08-31 by oe1wkl