oe1wkl/Morserino-32 #82: Echo Trainer: Wait time for response too long (if long inter-word space set)

beedaddy, 2023-10-16:
If you set the InterChar and InterWord spaces for the CW Academy beginner course (long pauses), the wait time on Echo Trainer is uncomfortably long.

If I see this correctly, this could be due to the variable `responsePause`, whose value is 5 by default.

For shorter InterWord spaces this is certainly useful, but in the given case probably a bit too long. Maybe the value should be adjustable via the morserino (or should probably adapt depending on the given interword space).

(Also, the time before the echo trainer starts with the first sequence is a bit long, which might have a different cause.)

oe1wkl, 2023-10-20:
This will also be fixed in the next version.

Willi

beedaddy, 2023-11-30:
Fixed with V5.1. Thanks!
cb83a00

closed 2023-11-30 by beedaddy