oe1wkl/Morserino-32 #87: pushbutton switch defective

dylanbeaudette, 2023-12-16:
Thank you for this wonderful kit. Everything works as expected except the red pushbutton switch is defective. Something is wrong with the mechanism or spring. How can I find a suitable replacement?

oe1wkl, 2023-12-17:
This is not an issue that can be dealt with on GitHub.

oe1wkl, 2023-12-17:
Please write an email to info@morserino.info. GitHub issues are for firmware bugs or feature requests.

closed 2023-12-17 by oe1wkl

dylanbeaudette, 2023-12-17:
Thanks. I couldn't find a suitable email on the website.

oe1wkl, 2023-12-17:
when you bought the kit, you got several emails, right? why not use that address??please mail your request to ***@***.*** meinem 🍎 Telefon gesendet

oe1wkl, 2023-12-17:
GitHub does not show email addresses.
I need your email  addressto je able to solve your problem...

write to.
info (at-sign) morserino dot info.

W

oe1wkl, 2023-12-17:
another option is to contact me through the groups.io morserino user group (which is the best way to get support, anyway).
W