oe1wkl/Morserino-32 #88: CW Acadamy sequence has changed

beedaddy, 2023-12-16:
In the latest CWA beginner guide (v4.2.7), the sequence has been altered. The sequence should now read:
**"teanois14rhdl25ucmw36?fy,pgq79/bv+kj80=xzBK.-ASNE@:"**

oe1wkl, 2023-12-17:
Will fix this in next release. Any idea why they pushed the comma forward?
Willi

beedaddy, 2023-12-17:
I think they have simply forgotten the comma so far (it was already part of the subsequent course and has not yet been introduced). They have also moved the letter Q forward by a few lessons and have introduced <AR>. It's a bit annoying that there are so many changes here. One could even think about making the sequence of letters configurable. Maybe that wouldn't be a bad idea at all.

closed 2024-01-25 by beedaddy