oe1wkl/Morserino-32 #93: Heltec Lora 32 VErsion 3

netnutmike, 2024-02-12:
I recently needed to replace my Lora32 board in one of my Moarserino 32s.  I ended up with a version 3 board.  I compiled version 5 and uploaded it to the V3 board no rpoblem but the screen is blank.

I am guessing that V5 might not be compatible with the Lora32 V3 board.  I got no errors on the compile or upload.

I cannot find V2 boards for 915 anyhere except alibaba.

oe1wkl, 2024-02-12:
V3 boards are not compatible with the Morserino-32, only V2.1 Heltec modules can be used.
You can get V2.1 modules (but only for 433 MHz) at the Morserino web shop.

Willi
OE1WKL

netnutmike, 2024-02-14:
Thank you for the reply.  I am in the US and trying to find a V2 board at 915 is pretty hard.

oe1wkl, 2024-02-14:
Is there a reason you need the 915 MHz version?
Licensed amateurs can use 430 MHz …

W

netnutmike, 2024-02-14:
That is true but to meet the letter of the law you have to ID in the data stream according to FCC rules.  Also, there is the possibility someone could use it that is not licensed.  No one would probbly ever notice or care.  I have 3 of them and the other 2 are using 915 so I would have to find the heltecs that came with the Morserino and put them back in the other 2.

oe1wkl, 2024-04-16:
As the protocol ahs been published, you can always ID with Morse code as ylou would with another transceiver. Re. I will have more 2.1 Heltec modules available in June.

closed 2025-07-09 by oe1wkl