oe1wkl/Morserino-32 #95: Problem mit dem lora TRX

DC8CH, 2024-04-15:
Im Modus LoraTRX blockiert der Empfänger bei Empfang und zeigt das empfangene Zeichen als Dauerlauf an. Es werden keine weiteren Zeichen empfangen. Senden funktioniert weiterhin allerdings ohne Anzeige.

Gibt es dafür Abhilfe?

Hardware ver1 
Software 5.1.1

beste Grüße 
Carsten

https://github.com/oe1wkl/Morserino-32/assets/161219167/e3192ec6-b4d3-41f5-9166-ba9840747cd2

oe1wkl, 2024-04-15:
So ein Problem wurde noch nie beobachtet... ev ein Hardwareproblem?Wieweit ist es reproduzierbar?WilliOE1WKLVon meinem Mobiltelefon gesendet

https://github.com/oe1wkl/Morserino-32/assets/161219167/e3192ec6-b4d3-41f5-9166-ba9840747cd2

DC8CH, 2024-04-15:
Dieses Problem tritt immer beim Empfang auf, wir haben schon probiert ob vllt Zuviel Leistung angekommen ist. Das war nicht der Fall für den Gegenvorschlag haben wir in einen Dummyload gesendet und auf den anderen Empfänger ebenfalls einen dumyload gesetzt. Der Fehler trat nur auf meinem moserino auf, der andere mit gleicher Firmware, aber hardware v2 klappt einwandfrei. Ansonsten funktioniert der morserino einwandfrei.Von meinem iPhone gesendet

https://github.com/oe1wkl/Morserino-32/assets/161219167/e3192ec6-b4d3-41f5-9166-ba9840747cd2

oe1wkl, 2024-04-15:
Sieht mir eher nach einem defekten LoRa Empfänger aus....WilliVon meinem Mobiltelefon gesendet

https://github.com/oe1wkl/Morserino-32/assets/161219167/e3192ec6-b4d3-41f5-9166-ba9840747cd2

oe1wkl, 2024-04-16:
Apparently not a software issue, hence closing this issue.

closed 2024-04-16 by oe1wkl

DC8CH, 2024-04-16:
Danke fürs prüfen! Von meinem iPhone gesendet

Closed #95 as completed.