oe1wkl/Morserino-32 #97: FR: configuration of lower vs upper case letters

rubienr, 2024-05-05:
I would appreciate a feature that allows to configure printing either lower or upper case letters for several reasons:

- on the small display i can hardly distinguish in-between `l`, `1`, `i` and `j`
- overall better readability with upper case letters
- faster learning for people memorizing visually i.e.

[image]

Umlauts, if not fitting in one line, could be still printed lower case.

oe1wkl, 2024-05-06:
First, obviously anyone can fork the PD code and modify it to their likings..
But I will not pursue this issue for the following reasons:
1. Learning code visually is the WORST IDEA ever, and you will never acquire a reasonable speed when you do it that way; any authorities on learning morse code agree that it has to be learned acoustically / orally, imn order to hardwire the rhythm into your brain. It is like learning music - you do not learn to play the piano by reading tabels that show where each tone is located on the keyboard.
2. The idea to use lowercase was done on purpose, the reason being that readability of lower case is actually a lot better then upper case . See for example https://en.wikipedia.org/wiki/All_caps and go to the section 'Readability'. Just try and read a page of "legalese" that at least in the US is written in all uppercase (I suppose in order to prevent people from actually reading it) - after half a page you will give up out of frustration.

rubienr, 2024-05-10:
**My apologies for striking a personal nerve.**  

I do respect your long CW experience but i disagree with your post since it arguments with far-fetched alternative facts. I would be perfectly satisfied if it just said _“I will not implement that because I personally dislike the visual method.”_.

This conservative tunnel-view is the reason why i didn’t learn Morse already 20 years ago (I will later go into details why). First of all I want to remind you that people are not equal. There are many learning methods out there. Visual and aural are just two of them. Ppl. usually learn how to learn, which by the way is the highest level of learning.  It would be ignorant to throw impaired, myope, dyslexic, poets, doctors, technicians, musicians, drummers etc. into one pot. Also, a serious and experienced developer wouldn’t dictate the use-case but rather ask the user for it.

Reading in detail the link you provided, it is clear that it does not apply 1:1 to the Morserino use-case. On the other hand, if you would want to search for material about why capitalized letters are not harder to read than lower case, you will find a lot. 

**Facts are**:

- Capitalized words, when reading full texts (like in news papers), are slower to read for the reason that the majority is not used to it. Otherwise capitalized letters are not significantly slower to read. But this is not the use-case here.
- capitalized letter + netiquette rules do not apply here
- lower case letters:
  - have more squiggly details that are harder to print on the low-res pixelated 128x64 display
  - symbol shape is not completely aligned with base/top line, hence requires more line spacing
- upper case letters:
  - always nicely fill the symbol's pixel-matrix (lesser pixelated)
  - require lesser line spacing
  - lesser pixelated squiggly details
  - larger (for myopic ppl.)
- all build-in radio decoders and software learning apps I have seen so far use capitalized letters, there must be a reasons:
  - Kenwood, 
  - Yeasu, 
  - Xiegu, 
  - morese-x, 
  - morseworld.com, 
  - MorseMania, 
  - MorseApp,  
  - MorseTrainer, ...
- majority of newspaper Headlines are still use capital lattes; there must be reasons too
- the 128x64 pixel UI would benefit from upper case letter(s) i.e. in echo mode; I am talking about the character(s) to repeat not the whole UI

I will not list more facts, since this would go beyond the scope of this discussion.

Now let me explain my experience (aural vs visual) before you demonize the visual method.
I tried to grasp the sound-feeling for letters but the main hindrance was that I couldn’t memorize the letters as dot-dash. I couldn’t link the sound to any letter and vice versa. I am also not the person who wants to pray dit-dahs in a church-chorus manner until exhaustion where someone has to tell me when to take a breath. If this is how Morse works, then: No, thank you!
With the [visual method](https://camo.githubusercontent.com/e1cf1757042f45345c5eac2e1b23934b877953f4f794805d3422caedd735e904/687474703a2f2f6c6561726e6d6f727365636f64652e696e666f2f77702d636f6e74656e742f75706c6f6164732f323031322f30352f6469746461682e706e67) I could memorize A-Z0-9 within 60 minutes before going to bed. The morning I could write texts with the paddle. As next step I memorized the [sorted Morse tree](https://user-images.githubusercontent.com/47689166/57795916-d0c1e300-774f-11e9-8989-7d2dc6b7e531.png) within two sessions (evenings) to improve my hearing. The rest of week I was training my hearing and special characters at [morsecode.world](morsecode.world). In the end, except of edge-cases, i don't have to recall either the visual method nor the tree. This is how i got ready for getting the feeling of each character's sound: almost effortless with sense of achievements and fun.

**Résumé**: 

There is not “the one method” and there is not “the one application” to learn Morse. I have gone through several Linux apps, Web apps, Mobile apps and Morserino. None of those is the mother of learning Morse.

On some unexpected occasion a few 8-years learned the visual method within minutes an we had fun half a day. I can’t imagine how I could give them an understanding or Morse with the aural/oral method so that they would be able to write and read full texts. At some point it has to make fun too.

PS: Note that i just wanted to constructively contribute to the project and asked whether or not it is possible to make the [decoder letters](https://github.com/oe1wkl/Morserino-32/blob/master/Software/src/Version%205/MorseDecoder.h#L35) configurable - not the whole UI. I also don't intend to start a religious discussion about the learning method. There is no such mother-of-morse-learnign-method!