oe1wkl/Morserino-32 #99: Probleme mit dem Einstellrädchen

Juergen2107, 2024-09-03:
Hallo,
Mein Einstellrädchen funktioniert plötzlich nicht mehr.
Was tun ?
73 de Juergen DL2JHK

oe1wkl, 2024-09-03:
Gar nicht? Oder nur der Drucktaster? Oder nur der Drehencoder?

Am wahrscheinlichsten ist, dass eine Lötstelle den Geist aufgegeben hat. Ich würde alle 5 Lötpads des Encoders gründlich nachlöten….

73

Willi

Juergen2107, 2024-09-03:
Danke Willi, ich glaube das ist es.Herzliche GrüsseJürgen --Gesendet mit der GMX Mail App

Juergen2107, 2024-10-09:
Dear Willi,

 

Es war wie du sagtest. Besten Dank !

 

Andere Frage:

 

Könnte man den Sensorkopf

 

 <https://www.bamatech.net/tastenzubehoer/sensorkopf-komplett.html> https://www.bamatech.net/tastenzubehoer/sensorkopf-komplett.html

 

irgendwie mit dem Morserino verbinden ?

 

Herzlichen Danke für deine Hilfe.

 

73 de Juergen 

DL2JHK, Berlin

closed 2024-10-27 by oe1wkl

Juergen2107, 2024-11-06:
> The BaMaTech sensorhead is working fine with the ESP32 touch pins: https://youtu.be/djfCpwT8nyQ

Dear Harald,

 

Many txs. How can I connect it with the Morserino ? Which pins on J1 and J2 do I have to use ?

 

73 de Juergen, DL2JHK

haklein, 2024-11-06:
Be aware that I've tested this with an ESP32-S3, the heltec v2 board on the Morserino is an older variant (with 8bit touch sensor instead of 16 bit). Still I'd suppose that it should work fine (it's just metal contacts for the touch pins after all..). You'd connect it to the touch pins as per schematics, PIN2 and PIN12 - https://github.com/oe1wkl/Morserino-32/blob/master/Documentation/Hardware/2nd_edition/schematics.pdf

Juergen2107, 2024-11-06:
Txs Harald, I’ll try to do so.

oe1wkl, 2024-11-07:
Somehow this is quite interesting, but unfortunately it is off topic ;-) keine Probleme mit dem Einstellrädchen, oder?

Juergen2107, 2024-11-07:
Nein, kalte Lötstelle beseitigt.Danke !!--Gesendet mit der GMX Mail App